openhab/org.openhab.binding.zwave #1698: [zwave] Aeotec Multisensor 7 binding

dandcent, 2021-12-11:
Hi
Can you please add binding for this device in the zwave database.
In openhab it is showing up as Z-Wave Node 020 (0371:0002:0018:1.1)

Here is the log
2021-12-11 07:29:24.494 [INFO ] [g.discovery.internal.PersistentInbox] - Added new thing 'zwave:device:8559c46ec1:node20' to inbox.

==> /var/log/openhab/events.log <==

2021-12-11 07:29:24.494 [INFO ] [openhab.event.InboxAddedEvent       ] - Discovery Result with UID 'zwave:device:8559c46ec1:node20' has been added.

==> /var/log/openhab/openhab.log <==

2021-12-11 07:29:29.599 [WARN ] [wave.discovery.ZWaveDiscoveryService] - NODE 20: Device discovery could not resolve to a thingType! 0371:0002:0018::1.1

2021-12-11 07:29:30.001 [INFO ] [ommandclass.ZWaveVersionCommandClass] - NODE 20: Command Class COMMAND_CLASS_BASIC has version 0!

==> /var/log/openhab/events.log <==

2021-12-11 07:29:50.833 [INFO ] [openhab.event.InboxRemovedEvent     ] - Discovery Result with UID 'zwave:device:8559c46ec1:node20' has been removed.

2021-12-11 07:29:50.841 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zwave:device:8559c46ec1:node20' changed from UNINITIALIZED to INITIALIZING

2021-12-11 07:29:50.858 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zwave:device:8559c46ec1:node20' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline

2021-12-11 07:29:50.871 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zwave:device:8559c46ec1:node20' changed from OFFLINE (BRIDGE_OFFLINE): Controller is offline to ONLINE

kaikreuzer, 2021-12-11:
Done!

cdjackson, 2021-12-11:
Please see the guide below for adding new devices to the database. You need to use the XML that OH generates to create the database entry. If you have any queries, please discuss on the OH forum where there will be more people who can help.

https://opensmarthouse.org/blog/zwave_device_database_guide

closed 2021-12-11 by cdjackson