openhab/org.openhab.binding.zwave #1700: FIBARO Door / Window Sensor 2 not recognized

smokku, 2021-12-16:
I just got two FIBARO's Door/Window Sensor 2, which are supported in OpenHAB like forever, but they do not get recognized.
Possibly some new version of the hardware?

* OH2 Version: 3.1.1 Release Build (openHABian)
* Binding Version: 3.1.1

If the problem is associated with a device, please provide the following -:

* Device name: FIBARO Door / Window Sensor 2
* Device version: FGDW-002(1) EU ver. 3.2 ZW5
* Link to [database](https://opensmarthouse.org/zwavedatabase/604):
* Link to the manual: https://manuals.fibaro.com/content/manuals/en/FGDW-002/FGDW-002-EN-T-v1.1.pdf

```
2021-12-16 17:19:19.619 [WARN ] [wave.discovery.ZWaveDiscoveryService] - NODE 37: Device discovery could not resolve to a thingType! Manufacturer data not known.

2021-12-16 17:19:19.628 [INFO ] [g.discovery.internal.PersistentInbox] - Added new thing 'zwave:device:6b61ef08d9:node37' to inbox.
```

cdjackson, 2021-12-16:
You need to wake the device up as the binding doesn’t know the manufacturer information so can’t correlate it with a database entry.

smokku, 2021-12-17:
If you mean waking by pressing the button, I did that.

smokku, 2021-12-17:
Now, after leaving it alone for a night, device appeared in my inbox.

cdjackson, 2021-12-17:
Maybe the device is broken then. From what I can tell, the binding is not communicating with the device, so it doesn't know what the device is. Of course I'm only going on very little information.

cdjackson, 2021-12-17:
Ok, cool, so I guess it woke up by itself :)

closed 2021-12-17 by cdjackson

smokku, 2021-12-17:
Looks like so.
```
Wakeup Interval
21600
```

Thanks for your help.