openhab/org.openhab.binding.zwave #1705: alarm_burglar channel state (for HomeSeer Floodlight Sensor HS-FLS100-G2) not updating when motion is detected

tyl0r, 2021-12-22:
Issue: The Alarmburglar item status is not updating when motion is detected. Does the database record need updating for this device? I'm having a similar problem described here: https://github.com/openhab/org.openhab.binding.zwave/issues/1112

Take a look at the OK.txt attachment below. It has an `Updating channel state` message whereas ALARM.txt does not. I think handleEvent() is returning null for `state` (which is causing the updating to be skipped) but I'm not sure:
https://github.com/openhab/org.openhab.binding.zwave/blob/3.2.0/src/main/java/org/openhab/binding/zwave/handler/ZWaveThingHandler.java#L1441

It looks like `alarm_burglar` channelType case (in the switch statement within the `handleNotificationReport` method) falls to the `default` case block. I'm not certain but I think the issue is somewhere in the `default` case block: https://github.com/openhab/org.openhab.binding.zwave/blob/dcc155dc7e2bb5c72d9170aa0b59be136b8a5b90/src/main/java/org/openhab/binding/zwave/internal/converter/ZWaveAlarmConverter.java#L362

I'm happy to test any changes.

tyl0r

* OH Version: 3.2.0
* Device name: HomeSeer Floodlight Sensor HS-FLS100-G2 
* https://github.com/openhab/org.openhab.binding.zwave/blob/c8e0b59c9ba058180ab4fbbaa3dbcbfb1e24b9c2/doc/homeseer/hsfls100g2_0_0.md

[ALARM.txt](https://github.com/openhab/org.openhab.binding.zwave/files/7758956/ALARM.txt)
[OK.txt](https://github.com/openhab/org.openhab.binding.zwave/files/7758957/OK.txt)

[network_f66344c2__node_5.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/7758722/network_f66344c2__node_5.xml.txt)

cdjackson, 2021-12-22:
I've changed this channel to a motion sensor - hopefully it will resolve this problem.

tyl0r, 2021-12-22:
Thanks Chris, I'll wait for the snapshot to build

tyl0r, 2021-12-24:
This works as expected. Thank you very much! 

I know there are probably other directions floating around somewhere but for anyone who cares to update their zwave binding, this is what worked for me. And you don't have delete all your zwave items (just the affected ones), and you don't have to re-exclude/include the zwave thing

- remove the zwave binding from the UI, then...
    ```
    ssh openhab
    cd /usr/share/openhab/addons
    wget https://ci.openhab.org/job/openHAB-ZWave/lastStableBuild/artifact/target/org.openhab.binding.zwave-3.3.0-SNAPSHOT.jar
    ```
  - (at this point your previously configured zwave things should show ONLINE still)
- delete your affected items and associated zwave things
- add new zwave Thing and hit Scan 
- add your equipment/items again