openhab/org.openhab.binding.zwave #1708: Use thermostat setpoint byte size and precision from capabilities report

kvolden, 2021-12-26:
The Zwave binding does not work properly with certain thermostats, because apparently it doesn't use the byte size and precision given by the thermostat through the Thermostat Setpoint Capabilities Get Command. In the case of HeatIt ZTRM3, it reports (and expects) a two byte setpoint, while it seems OH uses only one byte when setpoint is set below 25.5. In effect, the setpoint cannot be set to anything less than 25.5 degrees.

This was also discovered and fixed in OpenZWave, and it would be great if a similar patch could be made to the OH binding.

https://github.com/OpenZWave/open-zwave/pull/2458

OH version: 3.2.0
Zwave binding version: 3.2.0

olemr, 2022-01-09:
FYI. This is an analysis done by heatit:
[Test of thermostat Setpoint Command Class.pdf](https://github.com/openhab/org.openhab.binding.zwave/files/7834969/Test.of.thermostat.Setpoint.Command.Class.pdf)

olemr, 2022-07-23:
@cdjackson sorry to ping you, but what is your take on this?

cdjackson, 2022-07-23:
Sorry - there's nothing further than was discussed previously (as above) as I've not had the time to look at this.