openhab/org.openhab.binding.zwave #1720: Some things have wrong thing type definitions

J-N-K, 2022-01-29:
This was discovered when config validation was enabled in openHAB. One example is the `heiman_hs1sa_00_000`. It defines several config parameters named `group_X` of type `text`, but not all off them declare `multiple="true"` which leads to problems when the thing is initialized with a configuration that provides a list instead of a plain String.

cdjackson, 2022-01-29:
I think the problem rom the binding is (IIRC - I might be wrong) that the binding doesn't have the config description available - it only knows the values (ie the Configuration class is all that is passed to the handler). Maybe there's a way to get this easily - I've not checked and will try and take a look at this during the week if I can get a chance.

The other easy option is to configure all association groups for multiple=true - this might be the easier and safer option.

J-N-K, 2022-01-29:
I wonder how this can happen. It seems the thing handler is validating the configuration with the same code. How is it possible that it passes here and fails later? Or is there another way to update the configuration that I didn't see?

https://github.com/openhab/org.openhab.binding.zwave/blob/088efe5f138d2e99815ede57ee377536496936a7/src/main/java/org/openhab/binding/zwave/handler/ZWaveThingHandler.java#L593

cdjackson, 2022-01-29:
> I wonder how this can happen. 

You mean why it wasn't always validated? Fair point - I don't know.

One other question that has been burning in my mind (I've not followed everything that has been going on as it's only 7AM here) is why does saving the configuration fix the problem, but then restarting the binding (or OH) causes it to fail again. If saving it fixes the config and it validates, then shouldn't it also pass a startup validation?

cdjackson, 2022-01-29:
I've modified the database export to make all groups `multiple` so that should resolve this issue at least.

J-N-K, 2022-01-29:
If I understood correctly saving only fixes it, if you correct the wrong values.

My assumptions is

- Validation fails on initializing
- Values are corrected manually (and saved)
- Validation succeeds and thing initializes
- Thing updates configuration (this is NOT validated at the moment)

When the binding is restarted, the thing is disposed and re-initialized and we are back at the beginning.

J-N-K, 2022-01-29:
After looking at the code again: After the validation above the configuration is modified before it is finally updated. I guess that's are the invalid values (also the signed/unsigned and out-of-range) are introduced.

cdjackson, 2022-01-29:
Yep - good point.

Regarding signed/unsigned - this is a bigger problem to resolve so I'll likely do another "bodge" with the database export and in the exporter look for at least some instances where this can occur (eg a 1 byte value that has a max=255). The problem with this signed/unsigned issue is that the ZWave spec states all values are signed, so the binding manages the data in this way, but then manufacturers use unsigned values in their docs and this is hard to filter out.

petero-dk, 2022-01-29:
Fyi, downgrading core to 3.2.0-1 and running 3.3.0 addons - everything is fine again.

cdjackson, 2022-02-24:
Closing this as I believe it was fixed a few weeks ago.

closed 2022-02-24 by cdjackson