openhab/org.openhab.binding.zwave #1731: DimmableLight category seems deprecated

m4rcu5, 2022-02-01:
While tracing down a 404 on an icon in the openHAB interface, I noticed that my zwave dimmers got their category set to `DimmableLight` by default. Most likely due to the following statement.

https://github.com/openhab/org.openhab.binding.zwave/blob/4c6d5ba3d3be93144f8bbef0a4443daac7f76e57/src/main/resources/OH-INF/thing/channels.xml#L1263

While digging further to see why that icon is throwing a 404, it seems that the `DimmableLight` category is deprecated and removed from openHAB: https://github.com/openhab/openhab-core/pull/1915

Other bindings have acted upon that change by replacing `DimmableLight` by `Light`, see [hue](https://github.com/openhab/openhab-addons/pull/9335/files) and [milight](https://github.com/openhab/openhab-addons/pull/9336/files).

Should the zwave binding be updated as well?

lolodomo, 2022-02-15:
Yes  it must be changed. I submitted a PR last weekend for all bindings in the addon repo.

m4rcu5, 2022-03-01:
@lolodomo did you include the zwave binding as well in that PR?

lolodomo, 2022-03-01:
No Z-Wave binding was not included in my PR. As it is in a separate repo, a new PR is required.