openhab/org.openhab.binding.zwave #1735: Attempt to apply invalid configuration  (zwave binding)

lolodomo, 2022-02-16:


cdjackson, 2022-02-19:
The issue here is that the binding doesn't have access to the configuration, so it can only set the parameters to whatever the device reports. If the configuration is inconsistent with this, then the binding can't check this.

I would suggest to ensure that the database settings for the device in question are consistent with what the returns.

lolodomo, 2022-02-20:
Here is the log with the enhanced message:
```
13:13:16.661 [WARN ] [b.core.thing.binding.BaseThingHandler] - Attempt to apply invalid configuration 'Configuration[{key=zwave_version; type=String; value=1.55}, {key=group_5; type=EmptyList; value=[]}, {key=group_4; type=EmptyList; value=[]}, {key=group_6; type=EmptyList; value=[]}, {key=group_1; type=ArrayList; value=[controller]}, {key=group_3; type=EmptyList; value=[]}, {key=group_2; type=EmptyList; value=[]}, {key=switchall_mode; type=BigDecimal; value=0}, {key=wakeup_node; type=BigDecimal; value=0}, {key=zwave_devicetype; type=String; value=1}, {key=config_21_1; type=BigDecimal; value=50}, {key=wakeup_interval; type=BigDecimal; value=0}, {key=zwave_deviceid; type=String; value=3}, {key=config_1_1; type=BigDecimal; value=2}, {key=zwave_nodeid; type=String; value=2}, {key=config_26_2; type=BigDecimal; value=5}, {key=config_27_1; type=BigDecimal; value=7}, {key=config_3_1; type=BigDecimal; value=3}, {key=config_22_2; type=BigDecimal; value=2000}, {key=config_4_1; type=BigDecimal; value=0}, {key=config_23_2; type=BigDecimal; value=1000}, {key=config_24_2; type=BigDecimal; value=900}, {key=node_id; type=BigDecimal; value=2}, {key=zwave_manufacturer; type=String; value=357}, {key=config_25_2; type=BigDecimal; value=20}]' on thing 'zwave:nodon_msp31x1_00_000:controller:node2' blocked. This is most likely a bug: {config_24_2=The value 900 does not match allowed parameter options. Allowed options are: [ParameterOption [value="0", label="Metering Pulse disable"], ParameterOption [value="3600", label="Metering Pulse enable and send report every hour"], ParameterOption [value="21600", label="Enable and send report every 6 hours"], ParameterOption [value="43200", label="Enable and send report every 12 hours"]]}
```
It just confirms that the problem is with the parameter config_24_2 with value 900 not currently accepted.

>  would suggest to ensure that the database settings for the device in question are consistent with what the returns.

@cdjackson : do you mean that the option with value 900 must be added for this parameter ? If true, do you expect that I am proposing a PR for that or are you going to add it yourself ?

cdjackson, 2022-02-20:
I guess we answered this question in the other discussion :)

I'll close this - the change has been made to the database and I'm about to do a database export.

closed 2022-02-20 by cdjackson