openhab/org.openhab.binding.zwave #1739: Invalid file inteset/MC-V12.4_0_0.xml

lolodomo, 2022-02-21:
Using snapshot 2764, at startup I got this error:
```
20:29:23.312 [WARN ] [fig.xml.osgi.XmlDocumentBundleTracker] - The XML document '/OH-INF/thing/inteset/MC-V12.4_0_0.xml' in module 'org.openhab.binding.zwave' could not be parsed:
---- Debugging information ----
cause-exception     : java.lang.IllegalArgumentException
cause-message       : ID segment 'inteset_MC-V12.4_00_000' contains invalid characters. Each segment of the ID must match the pattern [\w-]*.
class               : org.openhab.core.thing.xml.internal.ThingTypeXmlResult
required-type       : org.openhab.core.thing.xml.internal.ThingTypeXmlResult
converter-type      : org.openhab.core.thing.xml.internal.ThingTypeConverter
path                : /thing-descriptions/thing-type
line number         : 120
class[1]            : java.util.ArrayList
required-type[1]    : java.util.ArrayList
converter-type[1]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
class[2]            : org.openhab.core.thing.xml.internal.ThingDescriptionList
required-type[2]    : org.openhab.core.thing.xml.internal.ThingDescriptionList
converter-type[2]   : org.openhab.core.thing.xml.internal.ThingDescriptionConverter
version             : 1.4.19
-------------------------------
com.thoughtworks.xstream.converters.ConversionException:
---- Debugging information ----
cause-exception     : java.lang.IllegalArgumentException
cause-message       : ID segment 'inteset_MC-V12.4_00_000' contains invalid characters. Each segment of the ID must match the pattern [\w-]*.
class               : org.openhab.core.thing.xml.internal.ThingTypeXmlResult
required-type       : org.openhab.core.thing.xml.internal.ThingTypeXmlResult
converter-type      : org.openhab.core.thing.xml.internal.ThingTypeConverter
path                : /thing-descriptions/thing-type
line number         : 120
class[1]            : java.util.ArrayList
required-type[1]    : java.util.ArrayList
converter-type[1]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
class[2]            : org.openhab.core.thing.xml.internal.ThingDescriptionList
required-type[2]    : org.openhab.core.thing.xml.internal.ThingDescriptionList
converter-type[2]   : org.openhab.core.thing.xml.internal.ThingDescriptionConverter
version             : 1.4.19
-------------------------------
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:81) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:72) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:68) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:52) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readBareItem(AbstractCollectionConverter.java:132) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:117) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:99) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:92) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:86) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:81) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:74) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:72) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:68) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:52) ~[bundleFile:1.4.19]
        at org.openhab.core.thing.xml.internal.ThingDescriptionConverter.unmarshal(ThingDescriptionConverter.java:53) ~[?:?]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:74) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:72) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:68) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:52) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:136) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1421) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1399) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1350) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1304) ~[bundleFile:1.4.19]
        at org.openhab.core.config.xml.util.XmlDocumentReader.readFromXML(XmlDocumentReader.java:105) ~[bundleFile:?]
        at org.openhab.core.config.xml.osgi.XmlDocumentBundleTracker.parseDocuments(XmlDocumentBundleTracker.java:396) [bundleFile:?]
        at org.openhab.core.config.xml.osgi.XmlDocumentBundleTracker.processBundle(XmlDocumentBundleTracker.java:382) [bundleFile:?]
        at org.openhab.core.config.xml.osgi.XmlDocumentBundleTracker$2.run(XmlDocumentBundleTracker.java:347) [bundleFile:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) [?:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:834) [?:?]
Caused by: java.lang.IllegalArgumentException: ID segment 'inteset_MC-V12.4_00_000' contains invalid characters. Each segment of the ID must match the pattern [\w-]*.
        at org.openhab.core.common.AbstractUID.validateSegment(AbstractUID.java:98) ~[?:?]
        at org.openhab.core.common.AbstractUID.<init>(AbstractUID.java:76) ~[?:?]
        at org.openhab.core.common.AbstractUID.<init>(AbstractUID.java:49) ~[?:?]
        at org.openhab.core.thing.UID.<init>(UID.java:48) ~[?:?]
        at org.openhab.core.thing.ThingTypeUID.<init>(ThingTypeUID.java:35) ~[?:?]
        at org.openhab.core.thing.xml.internal.ThingTypeConverter.unmarshalType(ThingTypeConverter.java:102) ~[?:?]
        at org.openhab.core.thing.xml.internal.ThingTypeConverter.unmarshalType(ThingTypeConverter.java:1) ~[?:?]
        at org.openhab.core.thing.xml.internal.AbstractDescriptionTypeConverter.unmarshal(AbstractDescriptionTypeConverter.java:194) ~[?:?]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:74) ~[bundleFile:1.4.19]
        ... 34 more
```

cdjackson, 2022-02-23:
I don't have that snapshot, but don't see this problem in the current version - is this still a problem or has it been fixed?

cdjackson, 2022-03-03:
@lolodomo - is this still an issue or do we close?

lolodomo, 2022-03-03:
I will install a new snapshot Saturday and will tell you if the problem persists. 
Normally it should, except if you changed something.

cdjackson, 2022-03-03:
I don't think anything has changed, but I also can't replicate, so if your system has this problem it would be good if you can take a look at it?

lolodomo, 2022-03-03:
Isn't the problem the presence of _ in the ID?

cdjackson, 2022-03-03:
Probably yes, but my point is that I can’t replicate the issue you report - I don’t see this in the database, and I don’t see this in the binding. So either someone fixed it at some point, or I don’t know why you have this problem.

lolodomo, 2022-03-03:
Do you think I should remove files in userdata/zwave?

cdjackson, 2022-03-03:
I don’t think the user data folder here should be an issue - there are no thing types stored there. You might want to remove the thing if it was created with this thing type (but I guess that’s not possible for it to be instantiated).

I think this must have been something that existed in one version, but was fixed at some point - I’m not 100% sure, but I don’t see this broken UID in the database now at least.

wborn, 2022-03-04:
> Caused by: java.lang.IllegalArgumentException: ID segment 'inteset_MC-V12.4_00_000' contains invalid characters. Each segment of the ID must match the pattern [\w-]*.

Looks like this was fixed with https://github.com/openhab/org.openhab.binding.zwave/pull/1740

lolodomo, 2022-03-05:
Tested with snapshot 2783 and the problem has disappeared.
Good.

closed 2022-03-05 by lolodomo