openhab/org.openhab.binding.zwave #1742: battery device (thing) is never offline despite removed battery for weeks.

TheTrueRandom, 2022-02-22:
* OH Version: openHAB 3.2.0 Release Build
* Binding Version: 3.2.0

If the problem is associated with a device, please provide the following -:

* Device name: FGSD002 Smoke Detector
* Device version: Don't know, bought a few month ago. 
* Link to [database](http://opensmarthouse.org/zwavedatabase): https://opensmarthouse.org/zwavedatabase/129  https://www.openhab.org/addons/bindings/zwave/thing.html?manufacturer=fibaro&file=fgsd002_0_0.html
* If the device is not in the database, provide a link to the manual.
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

<details>

```
<node>
  <homeId>0xc7837816</homeId>
  <nodeId>4</nodeId>
  <version>4</version>
  <manufacturer>0x10f</manufacturer>
  <deviceId>0x1003</deviceId>
  <deviceType>0xc02</deviceType>
  <listening>false</listening>
  <frequentlyListening>false</frequentlyListening>
  <routing>true</routing>
  <security>false</security>
  <beaming>true</beaming>
  <maxBaudRate>40000</maxBaudRate>
  <sleepDelay>1000</sleepDelay>
  <nodeInformationFrame>
    <commandClass>COMMAND_CLASS_ZWAVEPLUS_INFO</commandClass>
    <commandClass>COMMAND_CLASS_BASIC</commandClass>
    <commandClass>COMMAND_CLASS_VERSION</commandClass>
    <commandClass>COMMAND_CLASS_MANUFACTURER_SPECIFIC</commandClass>
    <commandClass>COMMAND_CLASS_DEVICE_RESET_LOCALLY</commandClass>
    <commandClass>COMMAND_CLASS_ASSOCIATION_GRP_INFO</commandClass>
    <commandClass>COMMAND_CLASS_ASSOCIATION</commandClass>
    <commandClass>COMMAND_CLASS_POWERLEVEL</commandClass>
    <commandClass>COMMAND_CLASS_WAKE_UP</commandClass>
    <commandClass>COMMAND_CLASS_BATTERY</commandClass>
    <commandClass>COMMAND_CLASS_ALARM</commandClass>
    <commandClass>COMMAND_CLASS_CRC_16_ENCAP</commandClass>
    <commandClass>COMMAND_CLASS_CONFIGURATION</commandClass>
    <commandClass>COMMAND_CLASS_SENSOR_MULTILEVEL</commandClass>
    <commandClass>COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION</commandClass>
    <commandClass>COMMAND_CLASS_APPLICATION_STATUS</commandClass>
    <commandClass>COMMAND_CLASS_SENSOR_ALARM</commandClass>
    <commandClass>COMMAND_CLASS_SECURITY</commandClass>
    <commandClass>COMMAND_CLASS_FIRMWARE_UPDATE_MD</commandClass>
  </nodeInformationFrame>
  <associationGroups class="concurrent-hash-map">
    <entry>
      <int>1</int>
      <associationGroup>
        <index>1</index>
        <maxNodes>0</maxNodes>
        <name>Lifeline</name>
        <profile1>0x0</profile1>
        <profile2>0x1</profile2>
        <commands>
          <commandClass>COMMAND_CLASS_SENSOR_MULTILEVEL</commandClass>
          <commandClass>COMMAND_CLASS_DEVICE_RESET_LOCALLY</commandClass>
          <commandClass>COMMAND_CLASS_ALARM</commandClass>
          <commandClass>COMMAND_CLASS_BATTERY</commandClass>
        </commands>
        <associations>
          <associationMember>
            <node>1</node>
          </associationMember>
        </associations>
      </associationGroup>
[... 529 lines cut ...]

## Possible Solution

## Steps to Reproduce (for Bugs)

1. Add the below configuration to openhab
2. See the device to be listed as online on openhab
3. Remove the batteries of the smoke detector
```
Bridge zwave:serial_zstick:controller "ZWave Controller" [ port="/dev/ttyACM1", security_networkkey="<hidden>" ] {
    fibaro_fgsd002_03_002 office "Smoke Detector Office" [ node_id=4, wakeup_interval=60 ]
}
```

## Context

I would like to know when a smoke detector is not reachable anymore.

## Your Environment

*Version used: (e.g., openHAB and add-on versions) openHAB 3.2.0 Release Build
*Environment name and version (e.g. Chrome 76, Java 8, Node.js 12.9, ...):
*Operating System and version (desktop or mobile, Windows 10, Raspbian Buster, ...): Raspberry Pi: Linux openhabian 5.10.63-v8+

sitenote: I created the issue on openhab addons https://github.com/openhab/openhab-addons/issues/12102 By now I put the batteries back. The XML is taken today, when the batteries are added again.

apella12, 2022-02-23:
I'm not the developer but have played around with zwave battery devices quite a bit. I can't say this is not an issue, but in your testing the device needs to be fully initialized before you pull the battery out and you can't restart OH or the binding during the entire time the battery is out. How to tell? when you pull the battery out the time stamp on that node XML file in user/zwave has to be after the controller time stamp and has to stay after the controller for the entire test period.

That said, there are posts in the forum where others have solved your requirement in other ways.  Again, not an expert, but I believe the expire binding was used.  I do not have anything critical like a smoke detector, but link a datetime item to my daily battery update and have displayed in the UI
[image]
Good luck

cdjackson, 2022-02-23:
I think this is a duplicate of 1065 so propose to close.

closed 2022-02-23 by cdjackson