openhab/org.openhab.binding.zwave #1758: Leviton ZW4SF Fan Speed Controller config parameter 7 will not update

smitopher, 2022-04-18:
Forum post https://community.openhab.org/t/openhab-3-3-milestone-discussion/132715/78

* OH2 Version: 3.3-M3
* Binding Version 3.3-M3

If the problem is associated with a device, please provide the following -:

* Device name: Leviton ZW4SF Fan Speed Controller
* Device version:
* Link to [database]https://opensmarthouse.org/zwavedatabase/1418:
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

If a device will not initialise, ensure the log is provided from the time OH starts.

Thank you.

smitopher, 2022-04-18:
My zwave log
[zwave.log](https://github.com/openhab/org.openhab.binding.zwave/files/8507049/zwave.log)

smitopher, 2022-04-18:
My node xml
[network_fad8d7d9__node_4.xml.txt]

smitopher, 2022-04-18:
My node xm
[network_fad8d7d9__node_4.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/8507073/network_fad8d7d9__node_4.xml.txt)
l

cdjackson, 2022-04-25:
Please can you summarise what the issue is. If the problem is parameter 7 not updating, please provide a log that shows you updating this parameter as I don't see any updates to node 4 in the log you provide. Or I might not understand the issue since there's limited information provided here.