openhab/org.openhab.binding.zwave #1770: New devices not showing up and previous device disappeared

fpap72, 2022-05-22:
Hello,

I recently added two devices to the database:

https://opensmarthouse.org/zwavedatabase/1479
https://opensmarthouse.org/zwavedatabase/1480

They were approved the 30th of April, 2022. I expected to have them available in the 3.3.0.M4 release, but they are not available, and they are unavailable in 3.3.0.M5 release as well. Why? I expected them to be added to the first release after the approval...

I also modified this device I added some months ago:

https://opensmarthouse.org/zwavedatabase/1434

and now it is no longer recognized by OpenHAB. Why?

This is what I see now of my node8, after updating to release 3.3.0.M4 and 3.3.0.M5. This node was perfecty recognized up to release 3.0.0.M3:

[image]

How can it happen that a device is REMOVED after a new release?

Thanks

mhilbush, 2022-05-22:
Those devices are in the main branch of the binding. However, they were committed to the code base 11 days ago on May 11. Therefore, they only will be in OH in builds since May 11.

Note that there often is some delay between when things are added to the database (or changed in the database) and when they show up in an OH build. Since M4 was built 21 days ago, and M5 was built 15 days ago, neither milestone release will contain these devices.

https://github.com/openhab/org.openhab.binding.zwave/commits/main/src/main/resources/OH-INF/thing/widom/edsc700_0_0.xml

fpap72, 2022-05-22:
Ok, thanks, I understand for the first two devices. But why the third one disappeared, after being available since release 3.2.0?

The only thing I can say is that I was modifying it in the same period when I was adding the other two devices, maybe this is a hint for understanding what happened.

mhilbush, 2022-05-22:
It's there now. However...

It was deleted in this commit on April 18. I don't know why...

https://github.com/openhab/org.openhab.binding.zwave/pull/1757/files#diff-95e2b23555e4dd46b5a97107d06f5376c751c99c02e3f8940f9612d868192463

Then it was added back in on May 11 in this commit.

https://github.com/openhab/org.openhab.binding.zwave/commit/cecb985a1e58f17440dd3aedcf14bb64f51fab4f#diff-95e2b23555e4dd46b5a97107d06f5376c751c99c02e3f8940f9612d868192463

fpap72, 2022-05-22:
Ok, thanks... Is there a way to have these three devices available now in OpenHAB? I think that if I switch to the snapshots I will have them, do you confirm?

Anyway, I'm not too much inclined to switch to the snapshots (I want to avoid as much as possible stability issues). Is there a way to update only the ZWave device database in the release I'm using now (3.3.0.M5)?

mhilbush, 2022-05-22:
Yes, you would need to install a snapshot build.

Or, you could uninstall the current version of the binding, then drop the latest zwave jar file into the addons directory. However, there wil be some dependencies that you will need to resolve manually. Unfortunately, I haven't done this in a very long time, so I don't remember what those dependencies are. You might find a post on the formus that explains how to do it.

Also, can you please close this issue, as there's nothing for Chris to do here?

fpap72, 2022-05-22:
Ok, thanks again!

closed 2022-05-22 by fpap72