openhab/org.openhab.binding.zwave #1778: Jasco 14318 3-way Light Switch changed

FishTuning, 2022-06-20:
OH Version  3.3.0-SNAPSHOT - Build #2957

I was running Main until 3 days ago when I switched to Snapshot to allow a new device to work. After installing this version the Jasco switch does not allow changing the value of Invert Switch(config_4_1). I had it set to "0" which has now become an invalid value. Changing it to 1 fixed the problem and the code window showed it set to 1. Changing it back to 0 seemed to work but the code window shows it set to null. Manually changing the code to 0 repeats the same error I had right after upgrading:

2022-06-19 00:21:25.048 [WARN ] [.core.thing.binding.BaseThingHandler] - Attempt to apply invalid configuration 'Configuration[{key=switchall_mode; type=BigDecimal; value=255}, {key=config_3_1; type=BigDecimal; value=0}, {key=group_1; type=ArrayList; value=[controller]}, {key=config_4_1; type=BigDecimal; value=0}, {key=group_3; type=EmptyList; value=[]}, {key=node_id; type=BigDecimal; value=26}, {key=group_2; type=EmptyList; value=[]}]' on thing 'zwave:device:2f49090b:node26' blocked. This is most likely a bug: {config_4_1=The value 0 does not match allowed parameter options. Allowed options are: [ParameterOption [value="1", label="Invert Switch"]]}

[network_ece4041c__node_26.txt](https://github.com/openhab/org.openhab.binding.zwave/files/8942794/network_ece4041c__node_26.txt)

apella12, 2022-06-20:
This is due to the new strict validation. (see this [thread](https://community.openhab.org/t/openhab-3-3-milestone-discussion/132715/51).  I can see in the that option 0 needs to be added to the jasco 14318 in the [DB](https://opensmarthouse.org/zwavedatabase/717).  Do you have DB write access?

FishTuning, 2022-06-20:
I thought I requested write access but I don't seem to have received it. I had it on the old site.

FishTuning, 2022-06-20:
And thank you for replying.

cdjackson, 2022-06-20:
If you had access on the old system then you need to request it again as I reset permissions when I transitioned the system a couple of years ago.

FishTuning, 2022-06-20:
I thought I did request it.

cdjackson, 2022-06-20:
Maybe you did - I really don't know and don't have any way to find out as I don't know what your username is.

FishTuning, 2022-06-20:
Username is Ira

cdjackson, 2022-06-20:
Thanks. You should have access now.

FishTuning, 2022-06-25:
So I have access now. When I look at the entry for this switch, it seems to be correct, though I notice the setting that's a problem has a red funnel and the one that works has a black funnel. What does that mean?

apella12, 2022-06-25:
I was not sure of your timing and wanted to try to get the fix in the 3.3 OH release (in case others have the switch- could avoid a forum issue or two), so made the change yesterday.  Hope that is okay. It is in the DB update 1779 that is approved to merge.

Red means that the limit to options is "True", black means it is "false". It could also have probably been fixed by changing the limit to options "false", but I left as "true" and added 0 as an option.

Bob

FishTuning, 2022-06-25:
No problem. I just looked at it and couldn't find anything that seemed wrong. It works fine the way it is once I set it to 1, but it's supposed to be set to zero. When I see it's fixed, I'll close it.

Thanks so much.

FishTuning, 2022-07-02:
Seems fixed in the last update I did.

closed 2022-07-02 by FishTuning