openhab/org.openhab.binding.zwave #1780: Zooz Zen32 broken

FishTuning, 2022-06-26:
Another something broken, though in this one I can't figure out what.  So I noticed 2 things.: 

A bunch of the settings have a red funnel which I think means the range is restricted.
In the thing page all the items with a red funnel show an X in the upper right corner of each parameter with a red funnel.
Changing them to "0" causes the X to vanish.
They are all configured with buttons giving me the choices 0-3
When I look at the database it looks like they are all set to allow values of 1-3

So, as hard as I try, something is happening I don't understand. Running snapshot updated last night.

I get this message when I try to configure it though I guess there's no useful information here :

[openhab.event.ConfigStatusInfoEvent ] - ConfigStatusInfo [
ConfigStatusMessages=[
ConfigStatusMessage [parameterName=config_20_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_21_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_11_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=group_3, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=group_2, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_17_4, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_8_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_18_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_9_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_12_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_16_4, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_19_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_3_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_13_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_14_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null], 
ConfigStatusMessage [parameterName=config_15_1, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null]]]

apella12, 2022-07-02:
This really doesn't belong here.  You would have much better and quicker support in the forum.  Primarily because someone there would likely have the device and there are quite a few folks with broad OH experience that are responding daily.  Also I'd note that your messages (above) are from the OH core, not the z-wave binding.  

That said, I noticed that a lot of channels were created for the Zen32 parameters (So a parameter can be changed by a rule, for example) in February 2022.  If you were on OH3.2 or earlier (working) and then jumped to OH3.3 M2 or later (broken), you could try to delete the Thing on the UI page (do not exclude), then go to inbox-zwave-scan to pick it up again (with the new channels- that look to have default values embedded)

If that works you can delete this issue.  If it doesn't work, I would still delete this issue and post it over in the forum for the wider audience and someone that has the device. (I don't, I'm just guessing)

Bob

cdjackson, 2022-07-02:
I would agree with @apella12, but I'll add the obligatory "please read the docs" here since it would be really good to get debug logging to help understand what the problem is. At the moment, there's really limited information and debug logs normally help.

However, it's also unclear to me what your problem is - you've listed a bunch of "stuff" - but what is the problem? Is the device not being configured? It seems from these status messages that the UI is sending the data to the binding, so all the comments about values and filters doesn't seem to be a problem - but again - I'm struggling to understand the problem.

Anyway, I'd really suggest to discuss this sort of issue on the forum - if there's really a bug, we can open it again.

FishTuning, 2022-07-03:
Today I pulled the Z-stick and deleted the device, did a factory reset, re-adopted it, corrected the node from 66 to 82 in items and relinked the orphaned items which seemed to fix the problem. Don't know what went wrong but it would seem it got misconfigured somehow during the upgrade to snapshot. I still have a problem with the scene buttons not working but I'm sure I figure that out soon enough.  Thanks so much for responding.  Sorry for posting in the wrong place but as it failed with the upgrade to snapshot which also broke a jasco switch I figured it was a similar problem. There's no obvious "beta" section on the board so I struggle with where or how to post what appear to be snapshot issues.