openhab/org.openhab.binding.zwave #1782: Hank RGB bulb HKZW-RGB01-V1.0 stopped working after upgrade from 3.2 to 3.3

robbert0h, 2022-06-29:
Hello,

Registering this for reference. I have been advised a db update can solve this and I'm intending to try to make such update.

After upgrading from OH 3.2 to 3.3, my AHank RGB bulb HKZW-RGB01-V1.0 stopped working:
`Thing 'zwave:device:512:node24' changed from UNINITIALIZED to UNINITIALIZED (HANDLER_CONFIGURATION_PENDING): {config_255_4=The value 1.0 does not match allowed parameter options. Allowed options are: [ParameterOption [value="1431655765", label="Resetting to the factory default"]]}`

This field is somewhat editable in the UI as a button/bullet that can be selected, but I don’t really fancy factory resetting the device. And I see no way to enter the mentioned value in there.

* Device name: Hank RGB bulb HKZW-RGB01-V1.0
* Link to: [database](https://opensmarthouse.org/zwavedatabase/598)

robbert0h, 2022-06-29:
From the manual it seems that the value of "1" (= default = do not factory reset) also should be allowed here, next to allowing the value 1431655765 (some kind of magic dword 0x55555555h) to trigger the factory reset.

[image]

cdjackson, 2022-06-29:
So yesterday I removed the restriction on other values (the limit to option box). This will fix the issue, but as per your comment above it might be better to add this "limit to options" back again and add the option for 1 as well.

I think I've just updated your access on the database so please feel free to do this.

robbert0h, 2022-06-30:
@cdjackson Thanks! I've now added the option for "1", enabled 'Limit to options' and removed the values for Minimum and Maximum, since we're not allowing free entry anymore.

Two questions (couldn't find answers to this in the documentation):
- The edited parameter is now showing a red sign in the parameter list - does this mean there is an issue with the new config? If so, how can I see what the issue is?
- I couldn't find the 'Request Review' button in the device overview. How do I request a review?

cdjackson, 2022-06-30:
> The edited parameter is now showing a red sign in the parameter list - does this mean there is an issue with the new config? If so, how can I see what the issue is?

No - it just means that the "limit to options" flag is true

> I couldn't find the 'Request Review' button in the device overview. How do I request a review?

I think I already requested review, so the button won't be shown as it's already in the review state

robbert0h, 2022-07-03:
@cdjackson This is now solved in the latest nightly build of the binding, so this issue can be closed.

closed 2022-07-11 by robbert0h