openhab/org.openhab.binding.zwave #1795: YRD446  will not initalize due to misconfig "The value 0.0 does not match allowed parameter options."

billfor, 2022-08-11:
No problems with this device on openhab 2.x to 3.2, until updating yesterday to 3.3.

HANDLER_CONFIGURATION_PENDING
{config_15_1=The value 0.0 does not match allowed parameter options. Allowed options are: [ParameterOption [value="1", label="Factory Reset"]]} 

In order to ensure we have the best information to deal with the issue, please fill in the following information (delete as necessary depending on the issue).

* OH Version: 3.3 
* Binding Version: org.openhab.binding.zwave-3.4.0-20220809.034710-13.jar

If the problem is associated with a device, please provide the following -:

* Device name: Z-Wave Node 005: YRD446 Yale Real Living Key Free Touchscreen Assure Lock with Z-Wave and Bluetooth

* Link to [database](http://opensmarthouse.org/zwavedatabase):  https://www.opensmarthouse.org/zwavedatabase/624

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](https://www.opensmarthouse.org/utilities/logviewer/zwave/) to see if you can identify the problem.

If a device will not initialise, ensure the log is provided from the time OH starts.

Thank you.

billfor, 2022-08-11:
Falling back to the 3.2 binding does not fix the problem.

[network_ef1de3db__node_5.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/9305938/network_ef1de3db__node_5.xml.txt)

cdjackson, 2022-08-11:
This is caused by changes that were made in the core a couple of versions ago. You either need to edit your configuration to change this to 1, or wait for the next release (hopefully in a few days time).

billfor, 2022-08-11:
Thank you Chris. I see that the modbus change in 3.3 probably broke some stuff. It is not urgent I just wanted to make sure you were aware.  I will read up on how to edit the configuration if I need it to work now.

cdjackson, 2022-08-11:
Any changes to modbus shouldn't impact the zwave binding. A big change was made to the core a little while back that added more checks to parameters in config files. This caused issues with a lot of bindings, and revealed some configuration issues in the database.

It should, hopefully, be fixed in the snapshot, and there should be a new milestone release this weekend.

billfor, 2022-08-15:
Confirm fixed in org.openhab.binding.zwave-3.4.0-20220815.034709-15.jar. Thanks Chris.

closed 2022-08-15 by billfor