openhab/org.openhab.binding.zwave #1803: Z-wave Scan option not showing up after PR 1798

apella12, 2022-09-15:
Don't know if you picked up on [this](https://community.openhab.org/t/things-sign-in-bottom-right-corner-z-wave-binding-not-listing/138605/6)
I have verified locally that reverting #1798 fixes the display problem.

There is a [workaround](https://community.openhab.org/t/things-sign-in-bottom-right-corner-z-wave-binding-not-listing/138605/9) to go directly, so maybe not too critical

cdjackson, 2022-09-16:
Sorry - I have limited access to internet at the moment - what is the problem? Ie can you please provide a description of the issue in here?

From the title, I don’t think I’ve made any changes that would impact this - unless the issue relates to the fact that the binding XML descriptor was renamed, but I assume @J-N-K did that for all bindings at the same time as he did the ZWave one, so the problem would be the same for all bindings?

apella12, 2022-09-16:
Sure.  Not super critical if you are still on Holiday.  I saw some DB updates, so thought maybe you were home.

The Z-wave binding option is missing After INBOX (+)
[image]

I'm running OH3.4M2 here with the stock Z-wave binding
  This was @J-N-K comment in the referenced chain.
[J-N-K](https://community.openhab.org/u/J-N-K)
[Jan N. Klug](https://community.openhab.org/u/J-N-K)
[Maintainer](https://community.openhab.org/g/maintainers)
[12d](https://community.openhab.org/t/things-sign-in-bottom-right-corner-z-wave-binding-not-listing/138605/7?u=apella12)
Could be. Probably it should be reverted and merged together with the core changes after 3.4 released. I can‘t do that.

cdjackson, 2022-09-17:
No - not home for another 10 days, but at least have internet access. bit more now that I'm in the UK rather than Africa.

Ok, I did ask before merging if this was a breaking change - I guess it was ;)

I'll try and revert the change.

J-N-K, 2022-09-17:
@cdjackson I guess I was mistaken there.

apella12, 2022-09-17:
My guess is that something in that part of the UI keys off the "binding" either in the directory or both to list the available options & "addon" is not recognized.  Anyway closing.

closed 2022-09-17 by apella12