openhab/org.openhab.binding.zwave #1818: donexon varmo TZ pro thermostat - sensor type 0x17 Water temp indication

markus7h, 2022-10-27:
The donexon thermostat is a danfoss derivate which include additional sensor information over the the original and other derivates like devolo etc. 

under the commend class COMMAND_CLASS_SENSOR_MULTILEVEL with sensor type 0x17 the water temp value can be read from the device

please include device specific detection for the donexon with additional water sensor channel
device type: slave
manufacturor id: 0x002
product type: 0x5fff
product id: 0xa010

cdjackson, 2022-10-27:
As far as I can tell this is already in the database.

https://opensmarthouse.org/zwavedatabase/295

markus7h, 2022-10-28:
you missed out, what I mentioned in my text. the devolo and others that are in the database ONLY have one temp sensor. this leads to a working device but with the lack of one temp channel. I do not know how to get an additional channel working if it does not come from the binding, but if there is a way, please let me know.

apella12, 2022-11-26:
So I think two things have to happen for this to work
1) `sensor_watertemperature` needs to be added to the DB drop down channel options. The following is already in the binding`WATER_TEMPERATURE(23, "WaterTemperature"),` , so I do not think anything needs to be done there? OR is there some other dropdown option that will pick this up?
2) @markus7h will need to add a new channel under the Sensor multilevel CC with that option in the community-maintained DB.  Read the blog to obtain write access.  

PS- I accidently modified the device while investigating the existing drop-down options but did get it back & marked for review.  Sorry

markus7h, 2022-11-26:
I did open a ticket to be able to make the suggested changes, let see how it turns out

apella12, 2022-11-26:
I realized on my afternoon run that my suggested dropdown might not work (or at least might have the temperature quantity).  I'm thinking a dropdown of `water_temperature[sensor_temperature]` might be an alternative based on the other entries.  Chris will be the only one that knows.

In support of all this, can you get a debug level file of the device sending that command.  In particular the hex string. Look for this section `31 ?1 17 ?2 ?3, etc.` 31 is Multilevel Sensor; ?1 is number of Bytes to follow; 17 is the water temperature ID; ?2 likely provides the size/scale/precision of the remaining Bytes; ?3, etc would be the value.  Also what is a typical water temperature for one of these radiators?  For instance 31 05 17 22 01 B8 would be 44.0 C

cdjackson, 2022-12-04:
> I did open a ticket to be able to make the suggested changes

I don't remember the ticket in question (I get a lot of them) but I guess this was actioned and you now have access.

> would the dropdown list of channel types over-ride any entry in the Config (type=) box

No - they are completely independent.

> Also thought maybe adding generic sensor_customdecimal on the dropdown could allow for obscure sensors on a more general basis.

As far as I'm aware, `sensor_customdecimal` is not a channel type, so it's not as simple as adding it to the dropdown. It will also require binding changes to the code to actually implement.