openhab/org.openhab.binding.zwave #1823: [Z-Wave Binding] Updating ZWave DB to get new devices

PleKy01, 2022-11-25:


apella12, 2022-11-25:
The way OH works is each release is a snapshot in time and is not normally (only if severe security issue found) remodified/updates after the release.  Your version is 3.3 (late June 2022) so any DB change after that time will not be retroactively added.  It is not a bug.  What you need to do is update to the latest snapshot binding that contains the device in question.

General outline

1)Remove UI Zwave binding - Don’t want two running
2) In karaf feature:install openhab-transport-serial - Needed in most systems, normally installed by UI
3) [Download 1](https://ci.openhab.org/job/openHAB-ZWave/) and place jar in addons (same folder as the openhab-addons-3.3.0.kar)
4) wait 30 seconds
Assuming you are on OH3.3 there should be no problem

PleKy01, 2022-12-09:
ok, thanks for the info.
According to the thread on community, it looks like it would be good to wait for 3.4 :(

Again Thanks for your time & work

apella12, 2022-12-10:
I tend to view the milestones as a better bet for stability. Plus there is a dedicated topic in the forum for milestone issues. The developers tend to cram a lot into the Release and a lot of issues get uncovered because many folks just wait for the release.  Yes, there is a code freeze, but the last milestones aren't around that long, so don't get any more testing than any ordinary milestone. Last release I thought 3.4M1 was "better" than 3.3.  Of course just my opinion.  I have 3.4M5 running on two systems without any issue.  If you aren't anxious to get the device working, waiting is an option