openhab/org.openhab.binding.zwave #1849: [zwave] ZW4SF Fan Speed Controller, joins network with invalid default config_7_1

switzer60, 2023-02-15:


apella12, 2023-02-15:
Out of curiosity have you tried changing the parameter to 254?  That is the default for the device in the binding. The reason it may appear as -2 is explained above.  If that doesn't work will need a fix to satisfy both the OH core rigorous validation and not confuse device owners (since the manual clearly states 254, 255, & 0 are the only options).  I have an idea on how but will need to run by the Zwave developer. However, the simplest is if setting the parameter as 254 works.

switzer60, 2023-03-21:
Yes.  You can change the value via the "Code" tab (but no using the "Thing" tab, perhaps a separate bug).  However, now I get this error:

2023-03-21 08:42:11.605 [WARN ] [.core.thing.binding.BaseThingHandler] - Attempt to apply invalid configuration 'Configuration[{key=config_7_1; type=BigDecimal; value=-1}, {key=config_3_1;
 type=BigDecimal; value=0}, {key=group_1; type=ArrayList; value=[controller]}, {key=config_4_1; type=BigDecimal; value=99}, {key=config_5_1; type=BigDecimal; value=0}, {key=config_6_1; typ
e=BigDecimal; value=3}, {key=node_id; type=BigDecimal; value=85}]' on thing 'zwave:device:04cb25f98c:node85' blocked. This is most likely a bug: {config_7_1=The value -1 does not match all
owed parameter options. Allowed options are: [ParameterOption [value="0", label="LED Off"], ParameterOption [value="254", label="Status Mode"], ParameterOption [value="255", label="Locator
 Mode"]]}

This is the Code from the Thing defintion:
UID: zwave:device:04cb25f98c:node85
label: "Z-Wave Node 085: ZW4SF Fan Speed Controller"
thingTypeUID: zwave:leviton_zw4sf_01_008
configuration:
  config_7_1: 254
  config_3_1: 10
  config_4_1: 99
  config_5_1: 0
  config_6_1: 3
  node_id: 85
bridgeUID: zwave:serial_zstick:04cb25f98c

apella12, 2023-03-21:
The message indicates you changed to 255? Is that true? Anyway one more test; set the config to "0" and see what happens. I suspect it will work fine until a DB fix can be made.

FYI, IMO - This is technically an issue with the OH core, not the zwave binding. The Zwave specification requires signed bytes.  What is probably happening is that when you change the value at 254, 255 or 0, the binding sends a SET message. The binding follows that up with a GET message and then the device sends a REPORT and the device REPORT is translated back to the signed byte value and put on the UI thing.  As noted above I think a change can be made to the zwave DB to get this working rather than bounce you back and forth.

airix1, 2024-08-24:
I am experiencing the same issue with both of my ZW4SF Fan controllers. Wondering if there is any update on this issue, or anything I can do to help?

apella12, 2024-08-25:
The vast majority of these have been resolved. (This thread is over a year old and probably should have been closed).  There is still an issue or two that pop up, usually in the forum because they get more attention there.  My guess this is parameter 7?

airix1, 2024-08-25:
Thanks for making the change!
I'll follow up once it test it.

Yes, selecting 0 for option 7 worked, but sometimes I would have to reselect if after a reboot, or the "handler error" would remain and the device wouldn't come online by itself.
Also this now allows textual configs to work for the device, since parameters cannot be configured.