openhab/org.openhab.binding.zwave #1871: Zwave binding does not come back online after hard reset

awaldfogel, 2023-05-23:
Set up the binding on OH 3.4 on a Pi4 with an Aeotec Gen5+ zwave stick.

Binding comes up and shows status ONLINE using either device /dev/ttyACM0 or udev symlink /dev/zwave. Expected behavior is that the binding reverts on reboot to its configured operation.

Problem: on a hard reset of the Pi, the binding does not come back online. Looked at the debug messages and the code, and it seems like the binding cannot reopen the device.

If I restarted on /dev/ttyACM0 and change the device to /dev/zwave then it restarts successfully.

However the failure for the binding to connect survives through OH restart and Pi reboot.

If I add a new udev symlink then the binding can connect to the device under the new name.

I don't know whether this is a problem with the binding, the OH serial library or the Java serial library, but it seems like a lock file is getting stuck and the binding could probably do something to clear out a stale lock.

For what it's worth, I have a UPS on the Pi, but as you know it's impossible to guarantee a computer will never hang or crash.  Forum response is mostly "Make sure your computer never crashes" 

Debug traces are posted on this thread:
https://community.openhab.org/t/openhabian-rpi4-zwave-gen5-falls-offline-on-restart/146694/17

cdjackson, 2023-05-24:
Please post the full logs here as the logs you reference aren't super helpful (pasted below for reference).

However it sounds like there's an issue with the udev links - the binding doesn't really have any control over these - it just tried to open the port that is defined - if the port isn't available, then it won't be able to open it, and this seems to be the case from your explanation above.

```
2023-05-18 15:07:04.459 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - Initializing ZWave serial controller.

2023-05-18 15:07:04.461 [DEBUG] [zwave.handler.ZWaveControllerHandler] - Initializing ZWave Controller zwave:serial_zstick:3070fe8bd9.

==> /var/log/openhab/events.log <==

2023-05-18 15:07:04.452 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing ‘zwave:serial_zstick:3070fe8bd9’ changed from UNINITIALIZED (HANDLER_MISSING_ERROR): Handler factory not found to INITIALIZING

2023-05-18 15:07:04.467 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing ‘zwave:serial_zstick:3070fe8bd9’ changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline

==> /var/log/openhab/openhab.log <==

2023-05-18 15:07:09.495 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - Connecting to serial port ‘/dev/zwave’

2023-05-18 15:07:09.528 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - Starting receive thread

2023-05-18 15:07:09.537 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Starting ZWave thread: Receive

2023-05-18 15:07:09.540 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - Serial port is initialized

2023-05-18 15:07:09.547 [DEBUG] [zwave.handler.ZWaveControllerHandler] - Initialising ZWave controller

2023-05-18 15:07:09.580 [INFO ] [ve.internal.protocol.ZWaveController] - Starting ZWave controller

```

awaldfogel, 2023-05-24:
I did a fresh openhabian install for a simple reproduction of what I'm seeing.
-> fresh install, turned off zram to eliminate file system cache issues. turned on zwave debug, inserted Gen5+ zwave and observed /dev/ttyACM0.  Zwave is online

`BRING UP AND INITIAL INSTALL OF ZWAVE BINDING - ONLINE @ 16:18:10.547 

2023-05-24 16:07:09.035 [INFO ] [.core.model.lsp.internal.ModelServer] - Started Language Server Protocol (LSP) service on port 5007
2023-05-24 16:07:13.230 [INFO ] [e.automation.internal.RuleEngineImpl] - Rule engine started.
2023-05-24 16:07:16.222 [INFO ] [ab.ui.habpanel.internal.HABPanelTile] - Started HABPanel at /habpanel
2023-05-24 16:08:08.382 [WARN ] [.io.rest.auth.internal.TokenResource] - Couldn't find a user with a session matching the provided refresh_token
2023-05-24 16:08:08.385 [WARN ] [.io.rest.auth.internal.TokenResource] - Token issuing failed: invalid_grant
2023-05-24 16:08:19.356 [INFO ] [p.auth.internal.AuthorizePageServlet] - First user account created: asher
2023-05-24 16:09:01.051 [INFO ] [ab.ui.habpanel.internal.HABPanelTile] - Stopped HABPanel
2023-05-24 16:09:01.123 [INFO ] [basic.internal.servlet.WebAppServlet] - Stopped Basic UI
==> /var/log/openhab/events.log <==
==> /var/log/openhab/openhab.log <==
2023-05-24 16:12:10.649 [INFO ] [.core.model.lsp.internal.ModelServer] - Started Language Server Protocol (LSP) service on port 5007
2023-05-24 16:12:11.702 [INFO ] [e.automation.internal.RuleEngineImpl] - Rule engine started.
2023-05-24 16:12:19.424 [INFO ] [ab.ui.habpanel.internal.HABPanelTile] - Started HABPanel at /habpanel
2023-05-24 16:17:12.580 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : BundleComponentActivator : ComponentHolder created.
2023-05-24 16:17:12.585 [DEBUG] [g.zwave.internal.ZWaveHandlerFactory] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveHandlerFactory] : BundleComponentActivator : ComponentHolder created.
2023-05-24 16:17:12.589 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Dependency Manager created ConfigDescriptionRegistryinterface=org.openhab.core.config.core.ConfigDescriptionRegistry, filter=null, policy=static, cardinality=1..1, bind=setConfigDescriptionRegistry, unbind=unsetConfigDescriptionRegistry, updated=null, field=null, field-option=null, collection-type=null, parameter=null
2023-05-24 16:17:12.590 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Dependency Manager created ThingRegistryinterface=org.openhab.core.thing.ThingRegistry, filter=null, policy=static, cardinality=1..1, bind=setThingRegistry, unbind=unsetThingRegistry, updated=null, field=null, field-option=null, collection-type=null, parameter=null
2023-05-24 16:17:12.592 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Dependency Manager created ThingTypeRegistryinterface=org.openhab.core.thing.type.ThingTypeRegistry, filter=null, policy=static, cardinality=1..1, bind=setThingTypeRegistry, unbind=unsetThingTypeRegistry, updated=null, field=null, field-option=null, collection-type=null, parameter=null
2023-05-24 16:17:12.594 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Component created: DS=DS13, implementation=org.openhab.binding.zwave.internal.ZWaveConfigProvider, immediate=true, default-enabled=true, factory=null, configuration-policy=optional, activate=activate, deactivate=deactivate, modified=null configuration-pid=[org.openhab.binding.zwave.internal.ZWaveConfigProvider]
2023-05-24 16:17:12.596 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Component Services: scope=singleton, services=[org.openhab.core.config.core.ConfigDescriptionProvider, org.openhab.core.config.core.ConfigOptionProvider]
2023-05-24 16:17:12.597 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Component Properties: {}
2023-05-24 16:17:12.599 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Querying state disabled
2023-05-24 16:17:12.601 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Querying state disabled
2023-05-24 16:17:12.603 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Component can not be activated since it is in state disabled
2023-05-24 16:17:12.605 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Querying state disabled
[... cut ...]
2023-05-24 16:18:19.469 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 1: Node Init transaction completed with response COMPLETE
2023-05-24 16:18:19.470 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 1: Node advancer: FAILED_CHECK - Controller - terminating initialisation
2023-05-24 16:18:19.472 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 1: Node advancer - advancing to DONE
2023-05-24 16:18:19.473 [DEBUG] [l.initialization.ZWaveNodeSerializer] - NODE 1: Serializing to file /var/lib/openhab/zwave/network_c0ac376d__node_1.xml`

awaldfogel, 2023-05-24:
Hard reset and zwave does not come back  

`HARD RESET @ 16:20 - Initializing Zwave Serial Controller @ 16:22:48.792 but never comes ONLINE

2023-05-24 16:22:34.003 [INFO ] [.core.model.lsp.internal.ModelServer] - Started Language Server Protocol (LSP) service on port 5007
2023-05-24 16:22:38.243 [INFO ] [e.automation.internal.RuleEngineImpl] - Rule engine started.
2023-05-24 16:22:41.174 [INFO ] [ab.ui.habpanel.internal.HABPanelTile] - Started HABPanel at /habpanel
2023-05-24 16:22:41.226 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : BundleComponentActivator : ComponentHolder created.
2023-05-24 16:22:41.233 [DEBUG] [g.zwave.internal.ZWaveHandlerFactory] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveHandlerFactory] : BundleComponentActivator : ComponentHolder created.
2023-05-24 16:22:41.248 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Dependency Manager created ConfigDescriptionRegistryinterface=org.openhab.core.config.core.ConfigDescriptionRegistry, filter=null, policy=static, cardinality=1..1, bind=setConfigDescriptionRegistry, unbind=unsetConfigDescriptionRegistry, updated=null, field=null, field-option=null, collection-type=null, parameter=null
2023-05-24 16:22:41.251 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Dependency Manager created ThingRegistryinterface=org.openhab.core.thing.ThingRegistry, filter=null, policy=static, cardinality=1..1, bind=setThingRegistry, unbind=unsetThingRegistry, updated=null, field=null, field-option=null, collection-type=null, parameter=null
2023-05-24 16:22:41.253 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Dependency Manager created ThingTypeRegistryinterface=org.openhab.core.thing.type.ThingTypeRegistry, filter=null, policy=static, cardinality=1..1, bind=setThingTypeRegistry, unbind=unsetThingTypeRegistry, updated=null, field=null, field-option=null, collection-type=null, parameter=null
2023-05-24 16:22:41.256 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Component created: DS=DS13, implementation=org.openhab.binding.zwave.internal.ZWaveConfigProvider, immediate=true, default-enabled=true, factory=null, configuration-policy=optional, activate=activate, deactivate=deactivate, modified=null configuration-pid=[org.openhab.binding.zwave.internal.ZWaveConfigProvider]
2023-05-24 16:22:41.260 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Component Services: scope=singleton, services=[org.openhab.core.config.core.ConfigDescriptionProvider, org.openhab.core.config.core.ConfigOptionProvider]
2023-05-24 16:22:41.263 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Component Properties: {}
2023-05-24 16:22:41.265 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Querying state disabled
2023-05-24 16:22:41.268 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Querying state disabled
2023-05-24 16:22:41.271 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Component can not be activated since it is in state disabled
2023-05-24 16:22:41.273 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider] : Querying state disabled
2023-05-24 16:22:41.276 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider(305)] : Updating target filters
2023-05-24 16:22:41.279 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider(305)] : No change in target property for dependency ConfigDescriptionRegistry: currently registered: false
2023-05-24 16:22:41.281 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider(305)] :  No existing service listener to unregister for dependency ConfigDescriptionRegistry
2023-05-24 16:22:41.284 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider(305)] : Setting target property for dependency ConfigDescriptionRegistry to null
2023-05-24 16:22:41.286 [DEBUG] [g.zwave.internal.ZWaveConfigProvider] - bundle org.openhab.binding.zwave:3.4.4 (241)[org.openhab.binding.zwave.internal.ZWaveConfigProvider(305)] : New service tracker for ConfigDescriptionRegistry, initial active: false, previous references: {}, classFilter: (objectClass=org. [... cut ...]
==> /var/log/openhab/openhab.log <==
2023-05-24 16:22:48.792 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - Initializing ZWave serial controller.
2023-05-24 16:22:48.794 [DEBUG] [zwave.handler.ZWaveControllerHandler] - Initializing ZWave Controller zwave:serial_zstick:f5082285e3.
==> /var/log/openhab/events.log <==
2023-05-24 16:22:48.800 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zwave:serial_zstick:f5082285e3' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline`

awaldfogel, 2023-05-24:
Let me know if there are any other test you'd like me to run.  Install is on an RPi4 with the latest firmware.

awaldfogel, 2023-05-24:
FWIW I tried deleting 

/var/lock/LCK…ttyACM0

and the binding can restart. 

Apparently there's a script floating around to delete the LCK files on system start. Would it make sense for the zwave binding to delete lock files on initialization? I have no idea if the binding can see the absolute path or if there's some other risk from ignoring lock file status.

https://community.openhab.org/t/openhabian-rpi4-zwave-gen5-falls-offline-on-restart/146694/41

https://community.openhab.org/t/aeotec-z-wave-gen5-stick-stays-offline-after-container-restart-raspi-reboot/136990/19

cdjackson, 2023-05-24:
The binding shouldn't be deleting lock files - they are actually there for a reason.  I thought there had been a change to the driver to resolve the lock file issue, but in any case I think we can close this as it's not a binding issue.

awaldfogel, 2023-05-24:
Is there any valid reason for lock files older than current system uptime? I wonder if a middle ground is to delete lock files older than uptime?

cdjackson, 2023-05-24:
Again though - the binding shouldn't be trying to manage lock files. As I said earlier, I thought that there had been a fix to the lock file issue in the latest drivers, but this is in any case not something that bindings should be trying to resolve as COM ports are implemented differently for different environments.

awaldfogel, 2023-05-24:
totally agree the binding shouldn't have to do this. But we can't seem to get anyone's attention to resolve lock files in the drivers.

On the OH forum I'm getting pushback that recovering from a hard reset is a non-goal for the platform. But it's kind of hard to guarantee hard resets never happen IRL.

I'm running 3.4.4 which I think is up-to-date.

cdjackson, 2023-05-24:
I agree that the system should start properly from a hard reset, but also don't want to venture into the messy area of COM port implementations in the binding.

The binding doesn't know how a COM port is implemented - it's abstracted away by another OH serial port layer. So the binding can't know if there are lock files (windows doesn't use them at all) or where they are stored (different drivers put them in different places) so I don't think there's anything that can be done to resolve this in the binding without it being a big mess.

awaldfogel, 2023-05-24:
Was afraid there might be some problem with too much abstraction.

Do you have any ability to poke the serial port maintainers? I'm implementing a workaround with a shell script in the init.d directory, but that feels like an even bigger hack.  I also think this issue is worth documenting in the binding because I'm neither the first nor last person who will trip over failure to reset

awaldfogel, 2023-05-24:
And before I forget: thanks for the work you're doing on the binding!

cdjackson, 2023-05-24:
The intermediate layer is part of the OH core, but that's basically just a middle layer to allow the core to change the serial driver without impacting on the binding, so it just defers to another library.

I don't mind if you want to add something to the doc of the binding. The other option is to add it to OH docs since ultimately this issue will impact all serial port users and possibly the best thing is to document it in the main OH docs, and then refer to that in the binding docs.

https://www.openhab.org/docs/administration/serial.html#serial-port-configuration

awaldfogel, 2023-05-24:
Is there a place a non-developer can open an issue with the serial core? 

I feel like I'm a little out of my depth since the symptom I observe relates to operation of the zwave binding and I don't want the core to just bounce it back to me as "User error".  I'm also hesitant to add something to the serial OH docs since I've only observed the lockup problem in this one place and don't have the bandwidth right now to search for broader anomalous serial behavior.

Unrelated but I had other issues with the serial binding. I couldn't pass control characters like "\n\r" and couldn't get any response on the OH forum, so I just gave up on the serial binding.

cdjackson, 2023-05-24:
> Is there a place a non-developer can open an issue with the serial core?

Sure - you can open an issue here -:

https://github.com/wborn/openhab-core/issues

You don't need to be a developer - it's the same as the issue you opened here

> I feel like I'm a little out of my depth since the symptom I observe relates to operation of the zwave binding and I don't want the core to just bounce it back to me as "User error". I'

I don't see why it would be bounced as user error. In any case, as discussed above, I don't think the binding has any way to manage the lock files, so discussing it in the OH core, where teh serial libraries are provided, seems to me the best place. Of course, they might also point you further down into the actual low level driver - I don't know.

> I'm also hesitant to add something to the serial OH docs since I've only observed the lockup problem in this one place and don't have the bandwidth right now to search for broader anomalous serial behavior.

Ok, that's fine. I was just suggesting adding your workaround regarding lock files to the docs.

awaldfogel, 2023-05-31:
No response from the OH serial core group.  Do you have any friends over there?