openhab/org.openhab.binding.zwave #1886: Eurotronics Sprit radiator thermostat is not initialising anymore

kristofejro, 2023-09-14:
Since last upgrade of OH to to Milestone 4.1 I can see that my Eurotronics Sprit radiator thermostat is not initialising anymore.
In the log file there is:

`2023-09-11 07:54:37.613 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zwave:device:711ae217:node54' changed from UNINITIALIZED (HANDLER_CONFIGURATION_PENDING): {config_2_1=The value 5 does not match allowed parameter options. Allowed options are: [ParameterOption [value="0", label="No Timeout LCD always on"]], config_5_1=The value 1 does not match allowed parameter options. Allowed options are: [ParameterOption [value="0", label="Unsolicited Temperature reporting disabled"]], config_6_1=The value 1 does not match allowed parameter options. Allowed options are: [ParameterOption [value="0", label="Unsolicited valve opening percentage reporting disabled."]]} to UNINITIALIZED`

I have already reported it on OH forum:
https://community.openhab.org/t/openhab-4-1-milestone-discussion/149502/9?u=kristofejro

OH Version: openHAB 4.1.0.M1 - Milestone Build
Binding Version: 4.1.0.M1

Database entry
https://opensmarthouse.org/zwavedatabase/710

As far as I can tell something was changed here:
https://github.com/openhab/org.openhab.binding.zwave/pull/1883

It would be great if someone could take a look on it.

apella12, 2023-09-14:
Maybe you did not see or understand my response in the forum??
`Don’t have the device, but noticed the “Limit to Options” was set incorrectly on these params. It was recently modified by someone on August 22. Fixed and marked for review. Should be in the next snapshot.`

What this means is the issue is fixed in the DB, just waiting for the next build.  Note that the DB entry is showing "Modified"

kristofejro, 2023-09-15:
Hi Bob,
Indeed I didn't understand your answer corectly. Even didn't know you are the maintainer :-)
Thanks.

closed 2023-09-15 by kristofejro

apella12, 2023-09-15:
Thanks for marking solved. 

FTR I am not the maintainer. The Zwave DB is community-updated by everyone that has access and _usually_ the device in question.  THE maintainer @cdjackson grants write access to the DB and then periodically reviews and updates the DB changes into the zwave binding.

I have seen your issue a number of times over the last two years (since changes were introduced in OH core that causes the issue, so knew exactly the problem. As I noted above, it was edited by someone else with write access that did not configure the parameter correctly.  That sometimes happens with so many "community" editors. I just edited parameters back to how they were prior to 8-22-2023 to save you the effort of registering and requesting write access. (and speed the process along). Normally I do not edit a device I do not have, but in this case what needed to be done was clear.