openhab/org.openhab.binding.zwave #1888: RaZberry 7 Pro not working properly

chris922, 2023-09-16:
* OH2 Version: 4.1.0.M1 (but issues exist even in previous versions)
* Binding Version: 4.1.0.M1

Log file (only zwave, DEBUG): [zwave.log](https://github.com/openhab/org.openhab.binding.zwave/files/12640530/zwave.log)

I always used the RaZberry directly on my Raspberry Pi (openHABian) and upgraded to the RaZberry 7 Pro some time ago. Unfortunately, this Z-Wave controller has some troubles. I can't say it is not working at all, it is more the case that randomly some devices are working and some are not. I even have some longer periods where all devices are working.
Right now it is again back to the "error state" where some devices are online, and some are offline:

[image]

It doesn't belong to a particular device model, e.g. "EG Jalousie 1 S" (node 14) and "EG Jalousie 1 SW" (node 15) are both exactly the same device models (Fibaro FGR222 Roller Shutter 2). The error message for node 14 is "Node is not communicating with controller":

I saw here are some PRs and issues around Z-Wave Gen7 in general so maybe it is just not yet supported.. but I am wondering why sometimes everything works like a charm.

Another strange thing I noticed: Restarting the device sometimes helps and sometimes not. Even though helped (maybe just by accident): Shutting down the device, removing the Z-Wave controller, waiting a few seconds, putting it on again, and starting the device and all devices are working again.

btw: I even exchanged the controller and the official z-way-server application seems to not have these kinds of troubles.

Looking forward to getting some help! If you have any ideas about what I can try or how I can better debug the issue let me know. I could even install some unreleased binding versions if you think that could help.

apella12, 2023-09-20:
Take a look at the log in the opensmarthouse log viewer utility. https://opensmarthouse.org/utilities/logviewer/zwave/ Your devices are pounding the controller with reports every few milliseconds.  Need to get that under control.  The good news is your controller still seems to be using 500 chip commands, so that is not a problem.

chris922, 2023-09-30:
Thanks @apella12, indeed it looks like some devices are going crazy. I reconfigure these spammy devices to send updates not that often... nevertheless this also doesn't solve the issues. I now switched back to my old RaZberry2 by doing a backup via z-way and restoring it. Using the RaZberry2 makes my z-wave network much more stable and I can again use my devices.
Thus it still looks to me like there are issues with the RaZberry 7 Pro.

Is there any update about the z-wave 700 standard for this binding? I mean.. this standard is already some years old and having looked at the other issues and PRs it seems it's not yet supported and there are still some things to do on this binding in this regard.

apella12, 2023-09-30:
I don't know what to say.  I'm not the maintainer, just try to help with the issues I can handle to help folks make progress. I did a 700 PR back in February, but it hasn't been commented.

LevinXX, 2024-09-20:
I'm facing a similar issue by transitioning from Razberry V2 to 7 Pro. Sometimes everything seems to be fine and fast. But sudden you will experience slow performance or devices even don't receive commands, but will later work again...
I didn't make any changes expect of changing the controller. There must something be wrong with it.
I'm on RPi4 and OH 4.2.1