openhab/org.openhab.binding.zwave #1897: Case sensitivity issue with doc/actec/zerodim2pol_0_0.md

lucasec, 2023-11-24:
My local git instance keeps showing that I have an unstaged change to the file `doc/actec/zerodim2pol_0_0.md`.

Was a little puzzled by this until I realized there are actually two versions of this file in the repo:
- `doc/actec/zerodim2pol_0_0.md` (lowercase `z`)
- `doc/actec/Zerodim2pol_0_0.md` (uppercase `Z`)

These both seem to have been introduced via a methodical database update by @cdjackson, so not sure if just submitting a PR to delete one of these files would be sufficient. The files are not quite exact duplicates of each other but I assume one is unintentional.

Obviously I could work around by doing development on a case sensitive filesystem, but may be worth fixing to avoid creating confusion for future contributors.

cdjackson, 2023-11-24:
Thanks for pointing this out.  I think removing the offending file will likely resolve this, but it does highlight an issue with the exporter because this obviously shouldn't have happened in the first place!

Leave it with me and I'll see if I can resolve this with an update to the export script.

apella12, 2023-11-24:
Thanks for fixing this.  I have struggled with this orphan file for months.  My bad for not creating an issue.

apella12, 2024-01-20:
My latest rebase of main popped up with a similar problem. Looks like it happened with #1914 with two files showing "modified" ( Capital Z)
```
doc\thermofloor\Zsmoke_0_0.md
src\main\resources\OH-INF\thing\thermofloor\Zsmoke_0_0.xml
```

but the lower case files appear to be in the binding
```
doc\thermofloor\zsmoke_0_0.md
src\main\resources\OH-INF\thing\thermofloor\zsmoke_0_0.xml
```