openhab/org.openhab.binding.zwave #1906: Please contribute to the database of supported USB sticks

andrewfg, 2023-12-16:
### Introduction

In OH core there is a PR [#3922](https://github.com/openhab/openhab-core/pull/3922) which on new OH installations does the following..
1. scan USB devices on the user's system
2. detect if any of them are a Z-Wave stick, and if so, 
3. suggest that the Z-Wave binding to be installed

This issue relates to point 2. above.

### Background

OH has several USB device scanners for different operating systems and configurations. These scanners can ALWAYS detect the USB stick `vendorId` and `productId` integer codes. Some (but not all) of these scanners can also retrieve the `manufacturer` and `product` description strings. But unfortunately, depending on the user's operating system and user rights, the latter strings may be missing.

Ideally in order to detect if a USB device supports Z-Wave we would search for common strings like "Z-Wave" in the `product` string. Yet on systems where the 'product' string is missing we need another way to identify what  the USB device supports.

In order to do this, I have started to create a database between hex 'vendorId:productId' pairs and the descriptions of known supported USB sticks. Example below..

```
            "0403:6001", "FTDI / UART", // this does not support Z-Wave or Zigbee
            "0403:8A28", "FTDI / ZigBee",
            "0451:16A8", "Texas Instruments / ZigBee",
            "0658:0200", "Aeotec / Z-Wave",
            "10C4:89FB", "Silicon Laboratories / ZigBee",
            "1CF1:0030", "dresden elektronik / ZigBee"
```

### Call to Action

Can you all please give me your contributions towards the above-mentioned database? Can you please provide me with the `vendorId`, `productId`, plus a brief description similar to the examples above? Many thanks for your contributions.

apella12, 2023-12-16:
Here is two that I use. (dmesg copy and then formatted).
```
New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
Product: Zooz_ZST10_700_Z-Wave_Stick
---------------
New USB device found, idVendor=1a86, idProduct=55d4, bcdDevice= 4.43
Product: 800 Z-Wave Stick
Manufacturer: Zooz
----------
"10c4:ea60", "Zooz_ZST10_700_Z-Wave_Stick",
"1a86:55d4", "Zooz 800 Z-Wave Stick"
```

Maybe could ask on the forum? (with instruction to run dmesg and pick out the sticks?)

marcelerkel, 2023-12-17:
Aeon Labs Z-Stick Series II
---------------------------
New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
Product: CP2102 USB to UART Bridge Controller
Manufacturer: Silicon Lab

This z-wave usb stick is pretty old. It is a z-wave (not z-wave+) stick.

andrewfg, 2023-12-17:
@marcelerkel interesting contribution thanks. It seems that this same underlying Silicon Labs hardware is used in the Zooz stick and the Aeon Labs stick.

sihui62, 2023-12-17:
`"0658:0200", "Sigma Designs, Inc. Aeotec Z-Stick Gen5 (ZW090) - UZB", //Z-Wave.me UZB Stick (https://z-wave.me/products/uzb/)`

andrewfg, 2023-12-17:
@sihui62 thanks for the contribution; apparently several producers use same chip from Sigma designs.

sihui62, 2023-12-17:
`"10c4:ea60", "Silicon Labs CP210x UART Bridge", //SONOFF Zigbee 3.0 USB Dongle Plus, based on TI CC2652P + CP2102N (https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/)`

sihui62, 2023-12-17:
> apparently several producers use same chip from Sigma designs

Yeah, looks like, which does it not make easier for you :-(

andrewfg, 2023-12-17:
> which does not make it easier for you

Actually it **_is_** (in general) easier. The suggestion finder works if the chip can be **uniquely linked** to a given binding communication system. i.e. we don't really care which manufacturer put their plastic case around the chip. ;)

However..
If a Z-Wave stick manufacturer would build their Z-Wave stick based on a **non unique** (generic) chip which is sometimes used for Z-Wave but is more commonly used for (say) memory sticks or whatever. This would lead to 'false-positives' that wrongly suggest installing a given binding.

The EnOcean system USB stick does use such a generic (UART) chip so it does cause false positives if discovery would only be based on the chip id alone. It is early days, but so far I did not get indications of this problem with Z-Wave or Zigbee sticks.  However we must definitely keep a sharp look out against such a case.

andrewfg, 2023-12-17:
You can check what is already in the database is [here](https://github.com/andrewfg/openhab-core/blob/discovery-usb/bundles/org.openhab.core.config.discovery.usbserial.otheros/src/main/java/org/openhab/core/config/discovery/usbserial/otheros/internal/UsbProductDatabase.java)

rkoshak, 2023-12-18:
> You can check what is already in the database [here](https://github.com/andrewfg/openhab-core/blob/discovery-usb/bundles/org.openhab.core.config.discovery.usbserial.otheros/src/main/java/org/openhab/core/config/discovery/usbserial/otheros/internal/UsbProductDatabase.java)

That link leads to a file not found

I have a Nortek HUSBZB-1 which is a combo Zwave (gen 5) and Zigbee controller. The Zwave appears first and Zigbee second (e.g. if no other devices are plugged in the Zigbee will be on ttyUSB0 (or ttyAMA0) and Zigbee appears on ttyUSB1. I'm not sure if that matters. The Zigbee is an Ember chipset. 

```
[    4.166839] usb 2-1: new full-speed USB device number 2 using xhci_hcd
[    4.322066] usb 2-1: New USB device found, idVendor=10c4, idProduct=8a2a, bcdDevice= 1.00
[    4.322094] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[    4.322109] usb 2-1: Product: HubZ Smart Home Controller
[    4.322128] usb 2-1: Manufacturer: Silicon Labs
[    4.322135] usb 2-1: SerialNumber: 312001DC
```

This one appears only once in the output for dmesg. Later on I see

```
[   23.564413] usbcore: registered new interface driver usbserial_generic
[   23.564424] usbserial: USB Serial support registered for generic
[   23.603471] usbcore: registered new interface driver cp210x
[   23.603485] usbserial: USB Serial support registered for cp210x
[   23.603506] cp210x 2-1:1.0: cp210x converter detected
[   23.605635] usb 2-1: cp210x converter now attached to ttyUSB0
[   23.605659] cp210x 2-1:1.1: cp210x converter detected
[   23.607496] usb 2-1: cp210x converter now attached to ttyUSB1
```

andrewfg, 2023-12-20:
@sihui62 just for the avoidance of doubt a few more questions:

> "0658:0200", "Sigma Designs, Inc. Aeotec Z-Stick Gen5 (ZW090) - UZB", //Z-Wave.me UZB Stick ZMEEUZB

1. Is the report above for two different sticks -- namely from Aeotec resp. Z-Wave.me ??
2. In your system do both sticks appear with the word 'Z-Wave' in their product description text?

> "10c4:ea60", "Silicon Labs CP210x UART Bridge", //SONOFF Zigbee 3.0 USB Dongle Plus, based on TI CC2652P + CP2102N

3. Is the report above for a Silicon Labs chip (vendor id 0x10c4)? Or for a Texas Instruments chip (vendor id 0x0451)?
4. In your system does it appear with the word 'Z-Wave' in its product description text?

sihui62, 2023-12-20:
> Is the report above for two different sticks

No, that is my one and only Zwave Stick. I have two USB Sticks attached to my server, one for Zwave (Zwave.me UZB), one for Zigbee (Sonoff Dongle).

> product description text?

Here is the complete output:

```
sihui@pve:~# lsusb
Bus 002 Device 004: ID 0781:55b0 SanDisk Corp. Portable SSD
Bus 002 Device 003: ID 0781:55b0 SanDisk Corp. Portable SSD
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0658:0200 Sigma Designs, Inc. Aeotec Z-Stick Gen5 (ZW090) - UZB
Bus 001 Device 003: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 005: ID 1b1c:0ab1 Corsair Flash Voyager
Bus 001 Device 002: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
> Is the report above for a Silicon Labs chip (vendor id 0x10c4)? Or for a Texas Instruments chip (vendor id 0x0451)?

As it says `10c4` I guess it is a Silicon Labs Chip ... or do I misinterpret you question?
The specs for that Dongle can be found here (pdf): 
[https://sonoff.tech/wp-content/uploads/2021/11/%E4%BA%A7%E5%93%81%E5%8F%82%E6%95%B0%E8%A1%A8-ZBDongle-P-20211008.pdf](https://sonoff.tech/wp-content/uploads/2021/11/%E4%BA%A7%E5%93%81%E5%8F%82%E6%95%B0%E8%A1%A8-ZBDongle-P-20211008.pdf)

andrewfg, 2023-12-20:
> As it says 10c4 I guess it is a Silicon Labs Chip ... or do I misinterpret you question?

It will be hard to auto suggest the Zigbee binding since the stick is based on the very common generic 0x10c4 UART chip, and there is apparently nothing in the name of that device which says 'Zigbee'. There is a trade off between making a positive suggestion for the Zigbee binding when the UART is used in a Zigbee stick versus the case when it is used in something else.

sihui62, 2023-12-20:
> EDIT: but from your dmesg listing it seems that id DOES have Zigbee in its name after all.

Yes, at least this one from ITEAD.