openhab/org.openhab.binding.zwave #1909: Shall we run mvn spotless:apply ??

andrewfg, 2023-12-26:
While working on a PR for this repository, I noticed that running 'mvn spotless:apply' results in many tens of files being reformatted.

As it is now common practice in openhab-core and openhab-addons to keep all source files spotless compliant, I wonder if we should do the same here? => If there are no objections, I would myself be quite happy to create a PR with spotless applied.

cdjackson, 2023-12-26:
This has been discussed in the past and I'm personally not keen as it looses a lot of history and IIRC from previous discussions, also changes the XML files which will be changed back with the next update.

andrewfg, 2023-12-26:
> changes the XML files which will be changed back with the next update.

From memory, it seemed to me that most of the format corrections were applied to java files and not xml files. Or??

cdjackson, 2023-12-27:
Ok, maybe I remembered that incorrectly, or possibly an exception was added to avoid changes. In any case the main point was I’d prefer not to loose the history and just do the updates as the files are touched rather than in one large reformat.

Nadahar, 2025-12-11:
It's quite painful as it is, because when you make a change to a file, and run spotless after editing (as I always do, because the spotless formatting isn't something that makes sense to be as a human, so I can't produce it manually), I end up having to revert changes to lots and lots of files that I didn't touch. A huge number of these files just have one or two indentation changes, so applying that to those files at least, wouldn't really hurt history much, and would reduce the number of files that must be reverted substantially.