openhab/org.openhab.binding.zwave #1935: Fibaro Relay FGS222 - add Channel for Device Configuration Paramters 4 and 5

robbelt, 2024-05-16:
I need the Device Configuration Paramters 4 (Relay 1: OFF-delay time (100ms steps)) and 5 (Relay 2: OFF-delay time (100ms steps)) for the Fibaro Relay Switch FGS222 as Thing Channels.

That is important for me because I use the Auto Off Function of the relays and they have to toggle EXACTLY after the specific time. I need to change that time often - that is why I need it as a Channel.
I tried to turn off the relay via rule after x milliseconds, but this is not acurate enough because of latencies.

* OH Version: openHAB 4.1.2
* Binding Version: 4.1.2

* Device name: Fibaro FGS222 Double Relay Switch 2x1.5kW
* Link to [database]: https://opensmarthouse.org/zwavedatabase/124

Thank you!

cdjackson, 2024-05-16:
I see you've registered on the database, so you can make the changes (thanks).

One point to note - I guess I'm a little surprised there's missing parameters on this device as it's been around a very long time. My concern is that you possibly have a newer version with these parameters, and if you add them to the existing database entry, it will cause problems for others if these parameters don't exist in older firmware...

So, please check the older docs to see if they existed, but just weren't in the database. If they don't exist in the older version, then we probably need to generate a new entry for the updated firmware version.

robbelt, 2024-05-16:
Oh nice. I added it to the Database and requested a review - hopefully in the right way :) 
Thank you!

cdjackson, 2024-05-16:
looks good - I'll push it through the system...

robbelt, 2024-05-16:
Thanks! I am absolute newbie here... so what are the next steps? When/How will it be usable for me?

cdjackson, 2024-05-16:
I'm doing a database export now. This will take maybe another 20 minutes or so (it's automated), then I'll need to merge that into the binding here on Github. Then at some point in the next 24 hours there will be a new snapshot built and you'll need to use that binding. You should be able to find out how to use the snapshot in the OH docs if I remember correctly.

closed 2024-05-18 by robbelt