openhab/org.openhab.binding.zwave #1949: Z-Wave controller goes offline when executing “Exclude Devices” from the controller’s Thing

lpn-ch, 2024-09-05:
Hello,

When I execute “Exclude Devices” from the controller’s Thing, the controller goes offline and comes back after a few seconds but it never enters the exclusion mode.

OH2 Version: 4.2.1
Device name: Zooz ZST39 800 controller
Device version: 1.20
https://www.getzooz.com/zooz-zst39-z-wave-long-range-usb-stick/

I include the DEBUG log file.

[Z-Wave log - Exclude devices.txt](https://github.com/user-attachments/files/16897024/Z-Wave.log.-.Exclude.devices.txt)

apella12, 2024-09-05:
FWIW _ I can duplicate this problem with OH4.3M1 in docker on Rpi3b with aeotec Zstick.  I ran the log here with the openhab core in debug and there does seem to be problems, but beyond my understanding.  The net effect is that the binding restarts rather than updates the parameter. 
[no-exclude-dev.txt](https://github.com/user-attachments/files/16899469/no-exclude-dev.txt)
 Also of interest is the UI code tab where the exclude parameter is now "true", but not functional.
[zstickconfig.txt](https://github.com/user-attachments/files/16899481/zstickconfig.txt)

brianlay714, 2024-09-05:
I don't recall it ever working.  I gave up on it a long time ago.

apella12, 2024-09-05:
Created a docker container on OH4.1.2 and it works fine, so somewhere earlier in 2024.
[oh4-1-2-exclude.txt](https://github.com/user-attachments/files/16899731/oh4-1-2-exclude.txt)