openhab/org.openhab.binding.zwave #1960: I need option to IGNORE selected (bad) nodes from z-wave controller

easypl, 2024-12-04:
My situation

Z-wave stick Gen5 (old) have some ghost nodes that cannot be removed via Z-wave PC controller.

The "isFailed" -> "RemofeFailed" cannot be performed (the z-stick hangs, no more blinkiing) so next step "remove" is not possible.
I did try EVERY possible way, no luck so far.
So I need to skip this node for quering by controller.

Same thing happend on Openhab start when device is quered on Z-wave binding start:

2024-12-04 02:01:16.530 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 112: Starting initialisation from EMPTYNODE

I would like to skip IGNORED nodes for ANY communication from controller to avoid controller freeze.

Is it possible?

easypl, 2024-12-04:
My OH ver is 4.2.3

apella12, 2024-12-12:
Not the developer, but a thought and options. This is new to me and with an older stick it could be the zstick.  I'm not sure modifying the binding, even if possible would make sense (IMHO).

Have you tried to back up the zstick, factory reset and restore the backup. With an old zstick you will have to use https://aeotec.freshdesk.com/support/solutions/articles/6000108806-z-stick-gen5-backup-software. Not the Silabs tool.

If you do buy a new Zstick I found this video helpful (You can't directly copy a network from a 500 to a newer 700/800 zstick. https://www.youtube.com/watch?v=lktIFOcg7GI