openhab/org.openhab.binding.zwave #1964: Nexa ZPR-111 not recognized

ahlund, 2024-12-31:
I have a Nexa ZPR-111 switch which is not recognized

* OH2 Version:4.3.0
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): Z-wave

If the problem is associated with a device, please provide the following -:

* Device name: Nexa Plugin on/off switch
* Device version:
* Link to [database](http://opensmarthouse.org/zwavedatabase):
* If the device is not in the database, provide a link to the manual. https://nexa.fra1.cdn.digitaloceanspaces.com/produkter/smartahem/zwave/zpr-111/manualer/86813_ZPR-111_manual.pdf
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](https://www.opensmarthouse.org/utilities/logviewer/zwave/) to see if you can identify the problem. 2024-12-28 16:50:26.224 [WARN ] [wave.discovery.ZWaveDiscoveryService] - NODE 11: Device discovery could not resolve to a thi
ngType! 0268:0002:1027::2.23

If a device will not initialise, ensure the log is provided from the time OH starts.

Thank you.

cdjackson, 2024-12-31:
It looks like this device is not in the database so will need to be added. Please [register](https://opensmarthouse.org/user/register-hidden.php) and request edit access so you can add it.

ahlund, 2024-12-31:
I know it's not in the register... ;-)

I will try...

ahlund, 2024-12-31:
However...
Your register link does not work in Safari...
[image]

cdjackson, 2024-12-31:
It works fine with Safari here

[image]

ahlund, 2025-01-05:
Ok, so how do I get edit access so I can attemt to get this working?

cdjackson, 2025-01-05:
Open a ticket and request access.