openhab/org.openhab.binding.zwave #1968: Edit access to device database

ahlund, 2025-01-05:
Hi!

I would like to request edit access to the device database. I have a Nexa ZPR-111 plug-in switch that I would like to try to get working. See [https://github.com/openhab/org.openhab.binding.zwave/issues/1964](url) for more information about the device.

apella12, 2025-01-06:
The process is to open a ticket after registering. If you have trouble check this post https://community.openhab.org/t/opensmarthouse-database-registration-down/159960/4

ahlund, 2025-01-07:
I am registered already, that´s why I opened this ticket...

apella12, 2025-01-07:
The ticket needs to be opened in the ZW DB, not here.
[image]

ahlund, 2025-01-07:
Aha! Sorry for the misunderstanding.

Thank You.

closed 2025-01-07 by ahlund